maximkulkin/esp32-homekit-camera #103: Can't apply patch in MacOS

Zhek1ch, 2022-09-28:
When using `git apply --directory="components/esp32-camera" esp32-camera.patch`:
I get this error, `error: patch failed: components/esp32-camera/driver/xclk.c:20 
error: components/esp32-camera/driver/xclk.c: patch does not apply`

When using `patch -p 1 < esp32-camera.patch`:
I get this, `Hunk #1 FAILED at 20.
1 out of 1 hunk FAILED -- saving rejects to file /Users/test/esp32/xtensa-esp32-elf/esp32-homekit-camera/esp32-camera.patch.rej`.

`esp32-camera.patch.rej` file:

*** 20,25 ****

      timer_conf.freq_hz = config->xclk_freq_hz;

      timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;

      timer_conf.timer_num = config->ledc_timer;

      esp_err_t err = ledc_timer_config(&timer_conf);

      if (err != ESP_OK) {
          ESP_LOGE(TAG, "ledc_timer_config failed, rc=%x", err);

--- 20,26 ----

      timer_conf.freq_hz = config->xclk_freq_hz;

      timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;

      timer_conf.timer_num = config->ledc_timer;

      +  timer_conf.clk_cfg = LEDC_USE_APB_CLK;

      esp_err_t err = ledc_timer_config(&timer_conf);

      if (err != ESP_OK) {
          ESP_LOGE(TAG, "ledc_timer_config failed, rc=%x", err);

How can I solve this?

arnaudbrst, 2022-10-22:
I have the same issue on Ubuntu 22.04

`error: patch failed: components/esp32-camera/driver/xclk.c:20
error: components/esp32-camera/driver/xclk.c: patch does not apply
`

I didn't find a solution yet

Numbski, 2023-03-02:
Same here.  Wonder what's up?

Numbski, 2023-03-02:
We might be able to revert driver/xclk.c to an earlier version.

Numbski, 2023-03-02:
https://github.com/espressif/esp32-camera/blob/8a365ee27240756cdd5a5e009db41ddd983b1e0e/driver/xclk.c

Using this version allows the patch to compile, but now we have a new error:

```
❯ make all                                                                                                                                                 ─╯
Toolchain path: /Users/tshadwick/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/tshadwick/esp32/esp-idf/requirements.txt are satisfied.
CC build/esp32-camera/driver/xclk.o
/Users/tshadwick/git/esp32-homekit-camera/components/esp32-camera/driver/xclk.c: In function 'camera_enable_out_clock':
/Users/tshadwick/git/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:23:15: error: 'ledc_timer_config_t {aka struct <anonymous>}' has no member named 'clk_cfg'
     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
               ^
/Users/tshadwick/git/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:23:26: error: 'LEDC_USE_APB_CLK' undeclared (first use in this function)
     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
                          ^
/Users/tshadwick/git/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:23:26: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [driver/xclk.o] Error 1
make: *** [component-esp32-camera-build] Error 2
```

Numbski, 2023-03-02:
I think the next course of action is going to be to run a diff between this (now patched) version and the current version in the repo.  There are some elements absent that need to be appropriately merged in.

Numbski, 2023-03-02:
If it is helpful:  old version but patched, versus current version unpatched (.backup):

```
❯ diff xclk.c xclk.c.backup                                                                                                                                ─╯
4a5
> #include "esp_system.h"
14c15
< esp_err_t camera_enable_out_clock(camera_config_t* config)
---
> esp_err_t xclk_timer_conf(int ledc_timer, int xclk_freq_hz)
16,18c17
<     periph_module_enable(PERIPH_LEDC_MODULE);
<
<     ledc_timer_config_t timer_conf = {};
---
>     ledc_timer_config_t timer_conf;
20c19
<     timer_conf.freq_hz = config->xclk_freq_hz;
---
>     timer_conf.freq_hz = xclk_freq_hz;
22,23c21,24
<     timer_conf.timer_num = config->ledc_timer;
<     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
---
> #if ESP_IDF_VERSION_MAJOR >= 4
>     timer_conf.clk_cfg = LEDC_AUTO_CLK;
> #endif
>     timer_conf.timer_num = (ledc_timer_t)ledc_timer;
25a27,37
>         ESP_LOGE(TAG, "ledc_timer_config failed for freq %d, rc=%x", xclk_freq_hz, err);
>     }
>     return err;
> }
>
> esp_err_t camera_enable_out_clock(camera_config_t* config)
> {
>     periph_module_enable(PERIPH_LEDC_MODULE);
>
>     esp_err_t err = xclk_timer_conf(config->ledc_timer, config->xclk_freq_hz);
>     if (err != ESP_OK) {
50d61
<
```

Numbski, 2023-03-02:
Could be as simple as switching LEDC_USE_APB_CLK to LEDC_AUTO_CLK?  I did try that, but it looks like timer_conf.clk_cfg is straight up lacking declaration in either case.  I'm a perl guy, and can do some javascript.  This is out of my depth.  Need the eyeballs of a c developer here.

Numbski, 2023-03-02:
Yeah, I'm done here.  There's no hint of that anywhere in the sources I've checked out, and I'm officially out of ideas:

```
❯ grep -R clk_cfg *                                                                                                                                        ─╯
Binary file build/bootloader_support/libbootloader_support.a matches
Binary file build/bootloader_support/src/bootloader_clock.o matches
Binary file build/driver/libdriver.a matches
Binary file build/driver/mcpwm.o matches
Binary file build/driver/spi_master.o matches
Binary file build/bootloader/bootloader_support/libbootloader_support.a matches
Binary file build/bootloader/bootloader_support/src/bootloader_clock.o matches
Binary file build/bootloader/bootloader.elf matches
components/esp32-camera/driver/xclk.c:    //timer_conf.clk_cfg = LEDC_USE_APB_CLK;
components/esp32-camera/driver/xclk.c:    timer_conf.clk_cfg = LEDC_AUTO_CLK;
components/esp32-camera/driver/xclk.c.backup:    timer_conf.clk_cfg = LEDC_AUTO_CLK;
esp32-camera.patch:+    timer_conf.clk_cfg = LEDC_USE_APB_CLK;
```

Numbski, 2023-03-02:
Had a moment of "ah ha!" this morning in that the struct for timer_conf is ledc_timer_config_t, and one would assume that is in the included driver/ledc.h.

Except:

```
❯ find . | grep ledc.h
```
Returns no results.  Seriously, I'm making an elementary c programming error here.  I'm certain of it.  If someone with even a little education on this subject looked, I'm certain you'd see my mistake and we'd be off to the races here.

Numbski, 2023-03-02:
That was the problem.  I added

```
#include </path/to/ledc.h>
```

Now we have a new sticking point.  The "right" way to do this will be to amend our include path.

```
❯ vi components/esp32-camera/driver/xclk.c
❯ make all
Toolchain path: /Users/tshadwick/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/tshadwick/esp32/esp-idf/requirements.txt are satisfied.
CC build/esp32-camera/driver/xclk.o
AR build/esp32-camera/libesp32-camera.a
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/component.mk:86: warning: undefined variable `IDF_TARGET'
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/component.mk:88: warning: undefined variable `IDF_TARGET'
CC build/homekit/src/debug.o
In file included from /Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:4:0:
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.h:33:23: error: unknown type name 'uint8_t'
 char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
                       ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.h:36:23: error: unknown type name 'uint8_t'
 char *text_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
                       ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.h:39:25: error: unknown type name 'uint8_t'
 char *binary_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
                         ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c: In function 'data_to_string':
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:8:12: error: implicit declaration of function 'data_to_stringv' [-Werror=implicit-function-declaration]
     return data_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:8:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c: In function 'text_to_string':
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:13:12: error: implicit declaration of function 'text_to_stringv' [-Werror=implicit-function-declaration]
     return text_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:13:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c: In function 'binary_to_string':
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:18:12: error: implicit declaration of function 'binary_to_stringv' [-Werror=implicit-function-declaration]
     return binary_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:18:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c: At top level:
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:22:24: error: unknown type name 'uint8_t'
 bool is_human_readable(uint8_t n, const byte **datas, size_t *sizes) {
                        ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:41:23: error: unknown type name 'uint8_t'
 char *text_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                       ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:79:25: error: unknown type name 'uint8_t'
 char *binary_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                         ^
/Users/tshadwick/git/esp32-homekit-camera/components/homekit/src/debug.c:109:23: error: unknown type name 'uint8_t'
 char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                       ^
cc1: some warnings being treated as errors
make[1]: *** [src/debug.o] Error 1
make: *** [component-homekit-build] Error 2
```

Numbski, 2023-03-02:
Add 

```
#include <stdint.h>
```
 to components/homekit/src/debug.h in order to get past this problem.

It builds MUCH further this time, but I hit yet another snag.

```
CC build/main/app_main.o
In file included from /Users/tshadwick/esp32/esp-idf/components/esp32/include/esp_timer.h:44:0,
                 from /Users/tshadwick/esp32/esp-idf/components/freertos/include/freertos/portmacro.h:82,
                 from /Users/tshadwick/esp32/esp-idf/components/freertos/include/freertos/portable.h:94,
                 from /Users/tshadwick/esp32/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from /Users/tshadwick/esp32/esp-idf/components/esp32/include/esp_wifi.h:62,
                 from /Users/tshadwick/git/esp32-homekit-camera/main/app_main.c:6:
/Users/tshadwick/git/esp32-homekit-camera/main/app_main.c: In function 'wifi_init':
/Users/tshadwick/git/esp32-homekit-camera/main/app_main.c:49:21: error: implicit declaration of function 'esp_event_loop_init' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
                     ^
/Users/tshadwick/esp32/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
cc1: some warnings being treated as errors
make[1]: *** [app_main.o] Error 1
make: *** [component-main-build] Error 2
```

Numbski, 2023-03-02:
Now I think I'm cooked, because we're into app_main.c, and now we have to determine the author's intent here.

*sigh*

barisonal, 2023-03-09:
Hey @Numbski I am facing same problems and I it became my new obsession :) So I saw that we have followed similar methods. So you are not alone!

I have been working on it for 4 days and I saw that it gives different type of errors when I change my installation method by using different versions of `esp-idf`. 

Let me share some critic points I saw;

- When I tried to build with [this instruction](https://www.studiopieters.nl/esp32-homekit-camera/), it was not building C code and giving the uint8_t type error. I found same solutions with you (by including `#include <stdint.h>` etc.) I realized that it's not a good way to modify source code. So this code should have been compiled correctly after the patch. 
- Instruction owner uses `esp-idf v3.2` but it looks like the current version is 5.0 in official website.
- I saw that someones tried to use `esp-idf v3.3` and `esp-idf v4.0` and I tried too.  I got different errors when I try to build with different version of esp-idf. 
- And I also saw your post which you say patch apply works with [this](https://github.com/espressif/esp32-camera/blob/8a365ee27240756cdd5a5e009db41ddd983b1e0e/driver/xclk.c) version of esp-camera. I tried and it worked for me too. That's the good point I think.

By the way, I have installed it successfully by modifying code base, including files etc. but after installation it didn't boot correctly. It was rebooting again and again when Homekit server starts and it was giving the error below; 

```
>>> HomeKit: Starting server
�[0;32mI (4414) esp32_camera: Initializing streaming�[0m
>>> HomeKit: Formatting HomeKit storage at 0x3a000
abort() was called at PC 0x400eba45 on core 1

Backtrace: 0x40093984:0x3ffc23e0 0x40093bad:0x3ffc2400 0x400eba45:0x3ffc2420 0x4008651b:0x3ffc2440 0x40193eca:0x3ffc2460 0x4019385c:0x3ffc2490 0x4008f6c5:0x3ffc2550

Rebooting...
```

So even we build and flashes to ESP32-CAM correctly, it is not working because we didn't fix the real problem. We just ensure it doesn't throw error. So, I think it's totally related with ESP-IDF and ESP-CAMERA versions. I will try to use different variations of library versions.