maximkulkin/esp32-homekit-camera #104: Compile error - esp_event_loop_init

codekobold, 2022-11-21:
`CC build/main/app_main.o
In file included from /home/pi/esp32/esp-idf/components/esp32/include/esp_timer.h:44:0,
                 from /home/pi/esp32/esp-idf/components/freertos/include/freertos/portmacro.h:82,
                 from /home/pi/esp32/esp-idf/components/freertos/include/freertos/portable.h:94,
                 from /home/pi/esp32/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from /home/pi/esp32/esp-idf/components/esp32/include/esp_wifi.h:62,
                 from /home/pi/esp32cam/esp32-homekit-camera/main/app_main.c:6:
/home/pi/esp32cam/esp32-homekit-camera/main/app_main.c: In function 'wifi_init':
/home/pi/esp32cam/esp32-homekit-camera/main/app_main.c:49:21: error: implicit declaration of function 'esp_event_loop_init' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
                     ^
/home/pi/esp32/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
cc1: some warnings being treated as errors
make[1]: *** [/home/pi/esp32/esp-idf/make/component_wrapper.mk:286: app_main.o] Error 1
make: *** [/home/pi/esp32/esp-idf//make/project.mk:530: component-main-build] Error 2`

esp_event_loop seems not to be defined in <esp_event.h>.

AchimPieters, 2023-02-09:
Do you use IDF5.0?

than-sad, 2023-02-23:
Hi AchimPieters, I'm using: `git clone -b v3.2 --recursive https://github.com/espressif/esp-idf.git;` Following you blog tutorial  [https://www.studiopieters.nl/esp32-homekit-camera/](url)

`/esp32/esp32-homekit-camera-maxim/esp32-homekit-camera/main/app_main.c: In function 'wifi_init':
/esp32/esp32-homekit-camera-maxim/esp32-homekit-camera/main/app_main.c:49:21: error: implicit declaration of function 'esp_event_loop_init' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
                     ^
/esp32/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
cc1: some warnings being treated as errors
make[1]: *** [app_main.o] Error 1
make: *** [component-main-build] Error 2`

Should I use IDF 5.0 or shouldn't I?

I'm little desperate trying to make it compile properly.

than-sad, 2023-02-26:
Ok I found in this git files changes just adding #include <esp_event_loop.h> after #include <esp_event.h> on file ./main/app_main.c solved the problem and allowed me to compile and flash it correctly FINALLY!!!

Now I'm fighting with no stream, just snapshot every 10 seconds.