maximkulkin/esp32-homekit-camera #109: Environment for successful build?

tophneal, 2023-02-20:
I've been struggling to get this to build across a few different devices (mostly on Fedora 37) with several versions of esp-idf (2.1, 3.2, 4.4, 5.0.1) and I'm constantly running into different errors.

I was wondering if anyone who has successfully built this project could share which version of IDF they used, and any modifications or changes that were needed.

than-sad, 2023-02-26:
Hi, just now I get it work after days in your same situation, I understand perfectly your frustration. I did it following this guide in Ventura 13.2 (Hackintosh) and High Sierra (Mac Mini Server). I have 0 knowledge of programing.

https://www.studiopieters.nl/esp32-homekit-camera/

After install esp-idf 3.2 following the guide I got the following compiling errors after setup the menuconfig:
`.In file included from /Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:4:0:
/Users/XXXXXXXXesp32-homekit-camera/components/homekit/src/debug.h:33:23: error: unknown type name 'uint8_t'
char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes);`

I solved adding: `#include <stdint.h>` as the first include inside ./components/homekit/src/debug.h

After that I got the following compiling error:
`../esp32-homekit-camera/main/app_main.c:49:21: error: implicit declaration of function 'esp_event_loop_init' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
                     ^
/Users/than_sad/esp32/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);`

And solver adding `#include <esp_event_loop.h>` after `#include <esp_event.h>` on file ./main/app_main.c

After that I was able to compile properly (just with some warnings) and flashed two esp32-CAM.

Now the problem I have is no streaming, just snapshot every 10 seconds and if I enter in streaming on Home app, esp32-CAM hangs and I have to reset it.

I hope this will help you at least to compile and add it to Home app.

Best regards, than.

than-sad, 2023-03-04:
Hi finally I got it with ESP-IDF 4.1.4 and mixing with Homekidd version. I have 320x240 streaming working in two ESP32-CAM.

I really don't remember all steeps I made and less the order... but if you try and try you will get it. Anyway ask me if you get bloqued.

barisonal, 2023-03-10:
Hello @than-sad, I was on the same status with you. I could finally make it work but without streaming, it gives just snapshot every 10 seconds.

When I replaced ESP-IDF with the v4.1.4 as you said, now it starts to give error at the end of the compiling.
`xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'
`

Did you get this error in your steps?

barisonal, 2023-03-10:
> Hi @barisonal, I never saw that error but looks related with xtensa, have you checked if there are residual folders from other installations, also wrong/old paths, make clean, etc.?. Or have you deleted all old esp-idf versiones and hidden folder .xtensa before try?
> 
> I found another user that get it work with esp-idf 4.4 but I don't know if streaming works for him, you can check here:
> 
> #96
> 
> Now I'm trying to compile this git with 4.1.4 but streaming fail with Guru meditation error.
> 
> The only one streaming worked for me is this (but isn't updated from 2020):
> 
> https://github.com/HomeKidd/esp32-homekit-camera
> 
> Try more and tell me how it is. Maybe we'll both make it!

I have checked the points you mentioned but everything looks good and yes I tried make clean etc.

I'm going to try HomeKidd version. It's my new obsession :)

barisonal, 2023-03-11:
Hello again @than-sad, I think it's totally related to combinations of the ESP-IDF, XTENSA and PYTHON versions. So let me know the Python version you used?

than-sad, 2023-03-11:
Hi! I used python 3.11.2 in MacOS Ventura (Hackintosh) and python 3.11.1 in macOS High Sierra (Mac Mini).
Also I have an association that run python3 when you execute python (to don't need to use python3 to run python files).

Actually I have esp-idf 4.1.4 in Ventura and esp-idf 3.2 in High Sierra, both can compile properly the esp32-homekit-camera from maxim and from homekidd.

I hope this will help you. Let me know your advances.

barisonal, 2023-03-11:
Hi again, thank you for the information.  I will have one more question to try to understand exact combination for me (and for everyone I hope :) )

You have 2 different setup and you are able to compile in both and flash to esp32-cam then everything works fine, camera streams video, not sends snapshot periodically right?

Because I cam compile without any error. But I saw that it does not work correctly on esp32-cam, it sends only snapshot 3-4 times and reboots itself.

than-sad, 2023-03-11:
Hi! The snapshot every 10 seconds is the normal behavior. If in your case after few snapshots the device hangs or reboot you should check with make monitor what’s happening in the device.

Answering to your question, yes I can compile both versions (maxim and homekidd) without errors and get snapshots without problems. In maxim version when I try the streaming I get guru meditation or no session ID errors and **don’t work**. With homekidd version I get streaming working but after minutes finally crash.

I want to get maxim version work because he updated the camera and other modules and maybe the performance will be better or at least more stable.

Anyway the streaming by default is set at 160x40 due to the divider stabilised at 4 (640x480/4) I changed to 2 getting 320x240 but with like 1 frame/second. Not useful for any security purpose.

Best regards.

barisonal, 2023-03-11:
Hi again! Okay then, it's more clear for me now. 

It's same here. It crashes and reboots with the `Guru Meditation Error: Core 0 panic'ed (LoadStoreError)` error when I try to get stream in Maxim version.

Now I have just tried Homekidd, it works but not stable, so it crashes with different type of errors randomly. It also stops streaming after a while. It was reported #51 in Maxim version and marked as closed. So it's weird.

By the way, resolution looks 640x480 in my Homekidd version with the setup below;

MacBook Pro M1 / macOS Ventura 13.2.1, Python 3.7.16, ESP-IDF v4.4

The other thing is the delay to start stream. It tries to connect about 10-15 second, sometimes 20 seconds even I'm in same network. I don't know is it expected behavior by the way.

barisonal, 2023-03-11:
> 

By the way I was getting same errors when I was trying with different versions of ESP-IDF and Python. But now with the versions you mentioned, it does not give these errors anymore and I don't need to add these lines.

than-sad, 2023-03-12:
Hi! That's great, you got it!

You say with Homekidd version you get 640x480 in snapshots or in streaming?

The delay for streaming is shorter in my case, around 3-5 seconds.

I hope @maximkulkin will come here soon or late and help us little bit to solve guru and session ID errors.

If you get any progress feel free to share it with me ;)

barisonal, 2023-03-12:
Thank you so much for your help and cooperation! :) Of course @maximkulkin should come here to help us...

Yes, I get 640x480 in snapshots. But 3-5 seconds is very good result, I will check it out to decrease delay on my side.

than-sad, 2023-03-12:
Sorry I think I didn't expressed myself properly. I get snapshots every 10 seconds (sometimes 10, sometimes 12, rare times 14). I think this is defined by Apple and can't be changed...

I though you mean the seconds that Homekit takes to start streaming, that's why I say 3-4 seconds.

Sorry for my misunderstood. And a pleasure to try to help. I felt desperate seeing no one answer me :P

barisonal, 2023-03-12:
Oh no you expressed correct :) I mean I will check it out to decrease delay to start streaming on my setup. So it currently takes 20 seconds to start streaming.

I totally agree with you for the snapshots durations. Apple decide the refresh times. It's same for me too.

than-sad, 2023-03-12:
Wow 20 seconds is a lot. If I found any information about it I will share with you.

Regards!

than-sad, 2023-03-13:
I want to share new discovered things not directly related with code. Maybe you know but in case not...

- If you compile like: `make -j flash monitor` you will compile with multicore Suport (really more faster)

- The serial baud rate by default `11520` in menuconfig Serial Flasher config is an old standard I supposed can be faster, and changed to `2Mbaud` getting the flashing process takes like 10 seconds.

With this two options the compile and flash process gets reduced drastically.

LazaroFilm, 2023-04-28:
have you had any chance getting this to work? 
I also get the preview image but when I request the camera stream, I get garbage data and the camera restarts. 
It also sometimes hangs when connecting to WIFI.

[image]