maximkulkin/esp32-homekit-camera #110: Compile Issue

walzen-shell, 2023-03-05:


than-sad, 2023-03-12:
Hi @walzen-shell. I can see two errors that I had:

`error: 'GPIO' undeclared (first use in this function)` In my case that happened when I didn't activated the GPIO option in menuconfig. Component config - Driver configuration - RTCIO configuration - Support array rtc_gpio_desc for ESP32 = check
`error: implicit declaration of function ...` In my case this happened because there is a function declaration without the corresponding #include

What esp-idf version are you using?

Regards.

walzen-shell, 2023-03-12:
I think the newest version: I just cloned the git repository from GitHub (https://github.com/espressif/esp-idf.git)

than-sad, 2023-03-13:
Hi @walzen-shell 

I got succeeded to compile with esp-idf 3.4 and 4.1 and @barisonal with 4.4. This code is old (around 2020) and a lot of things changed in esp-idf 5. I recommend you to try it with one of the versions I mentioned.

I think in esp-idf 5 that option was removed or at least changed of location. Im not an expert at all.

Regards.

walzen-shell, 2023-03-13:
Thank you
I tried now your suggested solution here https://www.studiopieters.nl/esp32-homekit-camera/ but I can't install esp-idf 3.2, because there is no install.sh or export.sh. How can I install an old version of esp-idf?
I used:
`git clone -b v3.2.3 --recursive https://github.com/espressif/esp-idf.git`
but there aren't those files in there

than-sad, 2023-03-14:
Hi @walzen-shell, 3.2.3 it's probably too old. Try deleting all esp-idf folders (including hidden ones if exist .espressif) and install the version 4.1.4.

Follow the tutorial just changing to: `git clone -b v4.1.4 --recursive https://github.com/espressif/esp-idf.git`

Test it and share results. Regards.

walzen-shell, 2023-03-14:
thank you @than-sad 
I get now following output in install.sh:
```
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /var/root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /var/root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /var/root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /var/root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /var/root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /var/root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/Users/hannesnagel/esp/esp-idf/tools/idf_tools.py", line 1459, in <module>
    main(sys.argv[1:])
  File "/Users/hannesnagel/esp/esp-idf/tools/idf_tools.py", line 1455, in main
    action_func(args)
  File "/Users/hannesnagel/esp/esp-idf/tools/idf_tools.py", line 1207, in action_install
    tool_obj.download(tool_version)
  File "/Users/hannesnagel/esp/esp-idf/tools/idf_tools.py", line 590, in download
    raise DownloadError()
__main__.DownloadError
```

than-sad, 2023-03-19:
Hi, the first one is related to certificates, you have to check your setup. The second one I don't know because I don't use docker.

Regards.

closed 2023-05-06 by walzen-shell